SebastianOberschwendtner/OTP15_oVario #22: Free UART for Bootloader

knuffel-v2, 2021-06-03:
# Expected Behavior

When booting with the bootloader by holding the *DOWN* button when plugging the micro USB in, the bootloader should only recognize
the USB port as a communication device.

# Current Behavior

The GPS module starts transmitting the NMEA stream immediately after powerup. Since the GPS uses an UART which is also actively monitored
by the bootloader, the bootloader tries to boot from the GPS module. :sweat_smile: The bootloader has an order in which it polls the
communcation interfaces and unfortunately the UART is polled before the USB ...

# Background Information

- The bootloader is placed inside the ROM of the microcontroller and can therefore not be changed. Its behaviour is more or less fixed.
- The GPS can be configured to change its output. This configuration can be saved into non-volatile memory of the GPS, but then this configuration
has to be changed at each power cycle.

# Possible Solutions

1. Disable the GPS when booting with the bootloader:

    This solution is a bit messy, but you could disable the LDO of the GPS module when booting with the bootloader. The LDO of the 
    GPS module has an enable pin, which can power off the GPS. This pin could be actively pulled high by a GPIO pin of the microcontroller during
    normal operation, but when booting with the bootloader this GPIO pin will be low and therefore the GPS disabled. A problem could be
    that the GPS fails during normal operation when flying, when something funny happens with this GPIO...

2. Use the *Debug-UART* for the GPS Module:

    The debug UART is *UART6* which is not monitored by the bootloader. This solution is cleaner, but you have to rewire the GPS module and
    change the code to use UART6. You also have to use a different DMA &rArr; *DMA2, Stream 3, Channel 5*. This DMA stream is still unused.

So what solution should we choose? :thinking:

knuffel-v2, 2021-06-03:
BTW: I've currently implemented solution **1**, which works so far. Further flight-testing required. ☝️

jkarpfinger, 2021-06-05:
Can you post a link to the stm bootloader specs/manual?

knuffel-v2, 2021-06-05:
Here you can find the application note [&rdca; AN2606](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwibta2um4HxAhXMMewKHR2HAKUQFjAAegQIBBAF&url=https%3A%2F%2Fwww.st.com%2Fresource%2Fen%2Fapplication_note%2Fcd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf&usg=AOvVaw2NOadXCCHepnPpgQX1P-ho) from *ST* which gives all the details about the bootloader.

---

We use the **Bootloader V3.x" of the `STM32F40xxx/41xxx` devices, which is described at page **109** of the application note 📜 .

---

In this application note, activity diagrams of the actions during boot are given. The following diagram is the one for our bootloader:

[image]
*taken from AN2606 from ST, p. 111*

jkarpfinger, 2021-06-06:
I'm currently changing the GPS to this [one](http://www.mateksys.com/?portfolio=m8q-5883).
Hacking into the enable pin of the LDO is not that easy and contradicts a easy to build thought.
So I'd rather go for solution 2. However I've not checked the power connectivity on the surrounding of the debug port.

knuffel-v2, 2021-06-06:
Allrighty. :+1:

According to your link, the GPS has a minimum input voltage of **4.0 V**. It probably regulates the input voltage internally to 3.3 V or 3.0 V, so going under 4.0 V would probably still work. I think the specs of our current GPS are quite the same.

&rArr; So we could think about connecting the GPS directly to the unregulated battery input. The GPS has its own LDO anyways and then the current through our "small" LDO would decrease as well. 🤔