SebastianOberschwendtner/OTP15_oVario #23: Ms5611: Baro reading is drifting

AliasJack, 2022-01-15:
# Expected Behavior

Steady pressure readings directly after startup. 

# Current Behavior

Pressure reading is drifting. 

# Failure Information (for bugs)

Apparently the ms5611 temperature is not read (value is set to 20 deg and doesn't change). 

## Steps to Reproduce

Switch on your vario. 

## Context

JK vario, no softbutton.

knuffel-v2, 2022-01-16:
Stranger things 🤔 

Does the bug also appear when using the firmware of this commit &rarr; 29244cd?

The ms5611 driver is not changed after this commit and the temperature readout does work with my version. I put my vario in the fridge, took it out again and turned it on. It warmed up from *13°C* (indicated in the Baro display page) to *25°C*. The change in the baro height during the temperature change was about 2m.

AliasJack, 2022-01-17:
The commit [https://github.com/knuffel-v2/OTP15_oVario/commit/29244cd16b3b923c84c481dd09b154a19e5b0c0f](url) works fine.

AliasJack, 2022-01-17:
Seems like an timing issue of the arbiter.

- If the program is started externally (debugging, uploading) everthing works fine. 
- If I do a powercycle, the baro temperature value is not read/updated.
- If I prolong the wait command in main->line95 from 100 to 300ms everthing works fine again.